KsenijaS/krakenx #1: Status

hurricanehrndz, 2017-07-21:
Can you please add a feature to report the status?
```
status = dev.read(0x81, 64)
liquid_temperature = status[1] + status[2]/10
print(liquid_temperature)
```

I'm sure array item 4 and 6 have something to do with RPM.

KsenijaS, 2017-08-05:
Hello,
Thank you for your input.
I tested your code and it doesn't calculate temperature accurately. For example:
Cam software shows that liquid temperature is 37 and according to your formula is 34.5.
Result is little bit better if temperature is calculated as status[1] + status[2] but it's still not correct.
Did you get same value of temperature as Cam software?

Jon0, 2017-10-02:
`status[1]` seems close to my sensors command output. (I don't have Cam software)
`status[2]` never seems to go above 10, it may be used as a decimal place.
`status[3] << 8 | status[4]` gives the fan speed rpm.
`status[5] << 8 | status[6]` gives the pump speed rpm.

hurricanehrndz, 2017-10-02:
Those were my findings as well, but I couldn't confirm because I don't have cam installed.

KsenijaS, 2017-10-04:
Hi,

I checked my sensors output and status[1] is close to CPUTIN value and Cam software may be buggy so I will add the feature to report status of temperature. 
But I am not sure about fan and pump speed: status[3] and status[4] are always 0 in my readings and also pump speed in sensors output is almost twice as when calculated by Jon's formula.  Also fan speed and pump speed in cam software never matched sensors output either.

Jon0, 2017-10-04:
Im using the Kraken x52 (1e71:170e) Maybe there is some variation in firmware?
I get a 17 byte result:
`status[0] = 0x04
status[1] = 0x1e
status[2] = 0x05
status[3] = 0x01
status[4] = 0xd0
status[5] = 0x04
status[6] = 0x66
status[7] = 0x00
status[8] = 0x00
status[9] = 0x00
status[10] = 0xff
status[11] = 0x02
status[12] = 0x00
status[13] = 0x01
status[14] = 0x08
status[15] = 0x00
status[16] = 0x00`

I monitored status[1] and status[2] while the temperature was increasing, and status[2] behaves like a decimal place, but never gets set to 0.

My fans are connected to the kraken controller. If your fans are connected to the motherboard then 0 values are expected.

The sensors rpm result comes from the motherboard chipset. For me it is unstable, jumping between 2000rpm and 6000rpm values. I dont think this value can be trusted.

The rest of the bytes 7-16 are usually the same one exception is status[15] gets set 0x1e after applying any change to fan or pump speeds.

The cam software may add some offset to any or all of these values.

KsenijaS, 2017-11-01:
Hi,

You're right my fan is connected to motherboard, I updated the code.

Thanks,
Ksenija

P.S Sorry for the late reply

closed 2017-12-27 by KsenijaS

Dids, 2018-03-27:
Can anyone confirm if these are still valid/up to date? I'm working with an X62 (0x1e71:0x170e), and while the temperature reading seems valid, the RPMs definitely don't, unless I'm not calculating them correctly.