KsenijaS/krakenx #9: Tai Chi effect

thevik, 2018-06-18:
Awesome program! you've saved me the overhead of running a windows VM to control my x62!

Is implementing the Tai Chi effect a possibility? It is the only effect that cam software offers that this program does not. 

Thanks!

KsenijaS, 2018-07-18:
Hi,

I think I have the newest CAM version and it doesn't include Tai Chi effect. I think I saw it in some previous version. If it comes up in some new version I will implement it.

Thanks

jneumaier, 2018-07-29:
Implemented in this [pull request](https://github.com/KsenijaS/krakenx/pull/14).

jneumaier, 2018-08-26:
@thevik You may test the Chaser / Tai Chi effect if you like to. You may also try Windows support without your VM.

closed 2018-09-17 by jneumaier