KsenijaS/krakenx #10: After issuing a color change, following commands fail to do anything.

thevik, 2018-06-24:
Arch Linux 4.17.2

After a fresh restart, with the LED's of my kraken x63 off:

colctl -m solid -c0 50,0,90  successfully changes the LED's from black to purple

immediately after I run:

colctl -m solid -c0 0,0,0 to change the LED's back to black but nothing happens.

no other colctl commands seem to work until I restart again.

willkill07, 2018-06-25:
I also report the same issue.

kernel version: 4.17.2-1-ARCH
Hardware: Kraken x62

jonasmalacofilho, 2018-07-25:
I can reproduce this as well.

jneumaier, 2018-08-26:
I guess this has been fixed accidentally. Works for me. Please verify.

jonasmalacofilho, 2018-08-27:
I can still reproduce this after a fresh boot, at least on Linux.

Could this maybe be related to detaching the kernel driver but never reattaching it?

jneumaier, 2018-09-05:
Thanks @jonasmalacofilho I merged the pull request #23 but only had time to verify everything works on Windows.
@thevik @willkill07 If you have the time, you are welcome to verfiy the fix on Linux. Otherwise I will just trust jonasmalacofilho's expertise and close the issue in a few days :-)

thevik, 2018-09-05:
Verified on Linux. Thanks for the fix! @jneumaier

jneumaier, 2018-09-05:
@thevik Perfect, thanks!

closed 2018-09-05 by jneumaier