KsenijaS/krakenx #11: Fan and pump speeds revert to defaults after a few seconds

jonasmalacofilho, 2018-07-17:
Issuing

```
# colctl --fan_speed 100 --pump_speed 100
```
makes the fans and the pump to spin up.

This can be confirmed both audibly and through `--status`:

```
# colctl --status
fan_speed 1786
liquid_temperature 28.6
pump_speed 2803
```

However, after approximately 10 seconds the fans spin down again, and so does the pump:

```
# colctl --status
Device status:
fan_speed 756
liquid_temperature 28.4
pump_speed 2239
```

I assume the firmware is reverting back to its defaults for some reason.  Have any of you experienced this already or can reproduce it?

Note: I'm running a brand new unit.  I'll later boot into Windows and post the firmware version.

jonasmalacofilho, 2018-07-17:
It looks like running the fans or the pump at "fixed" speeds requires sending regular updates.  CAM does that every 5 seconds or so.  Not sure if this was always the case or if is related to the recent firmware.

Still, profiles do (seem to) persist, even on this firmware.  The protocol for setting them looks simple, I will try to open a PR later.

jneumaier, 2018-08-25:
For me it behaves exactly like you descibed with firmware version 4.0.2. Pump speed resets to previous value after around 5 seconds. I guess it just regulates with the default silent mode curve. I found no information about how to edit theses curves in the protocol yet.

jonasmalacofilho, 2018-08-25:
I think it might be a failsafe: if CAM crashes, the cooler reverts to the configured profile.  From what CAM sets (and from what I have read the firmware ensures), the profiles will always ramp fan and pump speeds to 100% if the liquid reaches its maximum (un)safe operating temperature of 60°C.

An issue with profiles is that they default to 5°C steps (with no interpolation), which is kind of useless on the sensible 20-60°C range, and overkill elsewhere.  It is possible to set other steps, but I have experienced issues with the fan profile somehow interfering with the pump profile, and vice-versa.  CAM seems to bypass this (while it is running) by brute forcing and sending manual fan and pump speeds every second.

jneumaier, 2018-08-26:
Very nice! Would be perfect if flat was the default and you could optionally set a minimum and maximum pump speed with linear interpolation. Or maybe all values but I guess linear would be fine for a pump.

On the other hand interpolation between 0°C and 100°C does not really cover our use case :-) Maybe we would have to give some start/end temperatures for the min/max values as well. Do we have to cover this range or is it also possible to set less data points, e.g. 5 points between 30°C - 50°C? What do you think?

[Here](https://stackoverflow.com/questions/7343697/how-to-implement-linear-interpolation) are some interpolation alternatives.

jonasmalacofilho, 2018-08-26:
I absolutely agree with not using flat profiles: personally, I use a quite aggressive curve with the stock Aer P fans, for low noise during light loads (most of day), while still capable of handling the considerable peak heat output or even increased air temperatures.

Originally I tested a mechanism to automatically interpolate and adjust the temperature increments where it made more sense but, as I mentioned above, there were issues with one channel interfering with the other.  I think that the firmware only finds the appropriate table offset for the first channel, and reuses it for the other one, with the assumption that both will share the same temperature steps.

Since that failed, I decided to go back to interpolating with fixed 2°C increments between 20°C and 60°C.  I think this covers all use cases for those using dynamic fan or pump speeds; those exploring sub-ambient cooling or extremely low ambient temperatures will probably want to max out everything anyways.

That [has been working well](https://github.com/jonasmalacofilho/liquidctl) for me for the past week or two, and I think is ready to be contributed back to krakenx.  How does the the following interface look?

` ` `
colctl -fs (20,30),(30,50),(34,80),(40,90),(50,100) <pump profile> <lighting settings> 
` ` `

---

The reasons for another tool were both personal and technical.  Regardless, I will still contribute to krakenx in whatever way I can.

jonasmalacofilho, 2018-09-13:
@jneumaier I'm finally starting to work on the custom profiles for krakenx.

One thing that comes to mind, also considering #22, is that we could probably `import` liquidctl's driver here, as its API is modular enough to support different CLIs and use cases (other than the ones I implemented there).

What do you think?

KsenijaS, 2018-09-15:
Hello,
Sorry for the late reply. 
This sounds like a good idea, I seen your code @jonasmalacofilho and you have much better knowledge of protocols than any of us, also I would like you to add you as collaborator if you don't mind.
But in that case we should also update README file and I wonder if PROTOCOL.md should also be updated?

jonasmalacofilho, 2018-09-15:
I accepted the invitation, thanks.

Yes, I agree that we should update the README to reflect the change.  Not sure about PROTOCOL.md, maybe move it to liquidctl and keep it close to the implementation (besides, of course, updating it)?

jneumaier, 2018-09-17:
Fixed with #24. Thanks @jonasmalacofilho.
Feel free to close. Just leaving it open for README.md and PROTOCOL.md feedback for now. Maybe a slim protocol description poiting to the relevant source code parts would be enough. It is deep tech stuff any way and the source code explains it pretty well.

jonasmalacofilho, 2018-09-18:
Closing after moving the remaining bits to #27.

closed 2018-09-18 by jonasmalacofilho