KsenijaS/krakenx #15: Problem with installation in Windows

Carthago44, 2018-08-21:
Wenn I try to install krakenx, Python gives me the following:

**C:\Users\Toni\Downloads\ASUS\krakenx-master>python -m pip install krankenx
Collecting krankenx
  Could not find a version that satisfies the requirement krankenx (from versions: )
No matching distribution found for krankenx**

What do I make wrong? I am completely new with Phyton, installed actual version 3.7 with upgrading pip from 10 to 18.

Greetings,
Toni

jneumaier, 2018-08-21:
Oh, there is a typo in the docs. Should be krakenx not krankenx. But I did not check if the pip package has been updated. If this does not work, try the "Install from source" instructions.

Carthago44, 2018-08-22:
Ok, thank you! 
I found this mistake late in the evening yesterday (blind believing in drag and drop) and would tell this today, but I'm to late!

Installation of krakenx now works perfect, my actual problem is the colctl -command - I always get the message, that it is unknown.

jneumaier, 2018-08-22:
Yes, the maintainer of this project does not seem to have a lot of time currently. The public (pip package)[https://pypi.org/project/krakenx/] is more than a year old.

That is why i referred to the "Install from source" instructions.

Carthago44, 2018-08-22:
Many thanks for helping me Jens!
I'm quite new to Python and GitHub, so excuse my "stupid" questions (my actual programming is mostly in Lazarus-Pascal, Arduino and PLM51).
Two more questions:. 
- wherefrom do I have to start the installation from source - where is the right source directory?
- after installation: wherefrom can I use the colctl-Commands?

jneumaier, 2018-08-22:
No worries. For me it worked from the source folder (git clone & cd into cloned directory). Maybe it is because I use git bash as command line. You can also try to specify an absolute path instead of '.' at the end of the installation command. The colctl command should then be executable from anywhere in your path if your PATH environment variable includes the Python main and script folders. For me it worked out of the box with git bash, maybe you can try that. Feel free to ask for further support any time.

Carthago44, 2018-08-22:
Ok, Python already was in my local Path since installation. Now I tried to clone to GitHub desktop and a new installation from source should have been done, but there is no success in using colctl.
Here is a copy what happened:
 
C:\Users\Toni\Documents\GitHub\krakenx>pip install -e .
Obtaining file:///C:/Users/Toni/Documents/GitHub/krakenx
Requirement already satisfied: pyusb in c:\users\toni\appdata\local\programs\python\python37\lib\site-packages (from krakenx==0.0.1) (1.0.2)
Installing collected packages: krakenx
  Found existing installation: krakenx 0.0.1
    Uninstalling krakenx-0.0.1:
      Successfully uninstalled krakenx-0.0.1
  Running setup.py develop for krakenx
Successfully installed krakenx

C:\Users\Toni\Documents\GitHub\krakenx>colctl --help
Der Befehl "colctl" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

Carthago44, 2018-08-22:
Little update:

Colctl works in Git Bash, but nowhere else.

jneumaier, 2018-08-22:
Glad to hear it works partly. Please start the command line with a user account (not run as admin) and ensure both paths (main & scripts, e.g. ...\Python\Python36-32\Scripts) are in the **user** path environment. You should find a colctl file in this script directory. I had the same problem and also had the paths in the system environment which did not work for me. No idea why these PATH variables behave differently in system & user variables in a command line, but this at least fixed it for me.

jneumaier, 2018-08-22:
And Windows also does not make it easy to update your variables, maybe just restart or have a look [here](https://serverfault.com/questions/8855/how-do-you-add-a-windows-environment-variable-without-rebooting). In case you do not know what I mean: If you edit your environment variables, there are user variables in the top and system variables in the bottom part of the dialogue.

jneumaier, 2018-08-22:
Or was it the other way around?
In the docs I wrote
`Other accounts may need PATH updates for Python main and script folders in system environment (only available in user environment by default).` 
So I guess it was missing there, sorry. But got the exact same problem and had to add it there.

Carthago44, 2018-08-22:
Ok, the two paths are in the user environment at the moment. I will move them to system and make another try.
By the way: I've tested a little bit with my x62 fan speed using git bash and my default value after power on is about 1100 rpm. Using 60% as the lowest possible value gives me about 1700 rpm. So I think the default value after power on is lower than 60% - may be 50?

jneumaier, 2018-08-22:
Yes please report if you find out what is the correct configuration. I also struggled with the Windows command line and thought it was documented correctly.

The default if fspeed flag is not set is 30%, "Fan speed must be integer number between 25 and 100". 60% was for the pump and I wish this could be lower but is a firmware limit, no idea about the fan.

jneumaier, 2018-08-22:
Hm. I have it configured this way. But I guess it only makes sense for other user accounts like admin anyway. Do you have the colctl file in your "Scripts" folder? Does pip command work (pip itself without python -m, another exe in the scripts folder).
Currently it also works only in git bash for me. Maybe I remembered it wrong and gave up. Probably the shebang (#! first script line with interpreter path) simply does not work in Windows. Sorry for mixing this up and causing trouble for you. Probably it would only work with a batch file in Windows without a Unix Shell.
In Task Scheduler I use it like this: program "...\python.exe" and arguments "...\krakenx\bin\colctl -ps 60 --mode marquee --color0 9,33,71 --color1 2,7,15 --color_count 2 --animation_speed 0" where I directly use the cloned source. Maybe this helps you for your use case or you may create a batch file passing the arguments that way.

Carthago44, 2018-08-22:
Ok, back to user environment, but now even git bash doesn't help. Colctl is unknown.
I controlled PATH for typing mistakes, but everything seems to be correct. Making a cd to scripts where colctl is located helps nothing too. Unknown command …
Later on I will make use of task scheduler or maybe a batch file as you suggested, in the moment it would be nice to change pump and fan speed to find out my optimum values.

I think, I should stop my experiments for the moment - tomorrow is a new day and maybe there is more success.
Many thanks for your help for the moment; I will tell you, if there is something new.

Carthago44, 2018-08-23:
Some new results and answers to your questions:
- colctl only works with the two paths in user environment and executing it in git bash.
- colctl file is in my script folder
- pip works from anywhere
- testing parameters is possible using git bash; creating a task or a batch file should be possible.

Now I try to learn a little bit about python programming, how to edit source code and so on. In connection with GitHub this is very confusing for a beginner, but very interesting I think.

jneumaier, 2018-08-23:
Thanks for the update. It was just about Windows command line not compatible with a Unix shebang instruction, the .py file ending is required for Windows to make the script executable. I added a `colctl.py` redirection script to allow execution in Windows command lines.

It is not merged yet and you might try from my forked repository if you are interested:
https://github.com/jneumaier/krakenx

Also added some documentation, so I hope the issue becomes clear for new users.

You won't regret getting into git for sure. And Python is always nice at least for smaller scripting like we do in this project :-)

Carthago44, 2018-08-23:
I installed new from your source and now "python colctl -s" works fine in /bin directory.
Many thanks and sorry, I have no time for further testing in the moment - I think tomorrow I am able to report more.

jneumaier, 2018-08-23:
`colctl.py -s` might work anywhere now and in Windows command line.

Carthago44, 2018-08-24:
Ok, some new tests and their results:
- colctl only works in windows when starting it from a directory were it is in and in combination with    python, not standalone
- with git bash standalone operation from everywhere is possible
- pump speed 50 works fine; I changed the values from 60 to 50 in "my" source. Now 50 gives me the default speed which x62 has after power on (about 1100 rpm). Changing to 60 results in about 1700 rpm and back to 50 is 1100 again. 

At the moment I try to find out were cloning from GitHub, install with Python and so on locate all the project files - some are in documents/github/, others in python/source and in lib\site-packages - a little bit confusing.
Also confusing: there is pip, pip.3, pip3.7, easy_install, easy_install 3.7 and so on and as a result of installing my modified program I now get "e-1.4.5-py3.7.egg.info" instead of "kraken-0.0.2-egg.info" whatever this is and means.
Python and GitHub remember me to my first experiments with "Forth" for uCs like 68000 many years ago - nothing was like usual. Thanks to CAM software for learning something new!

jneumaier, 2018-08-24:
Yes colctl will not work in Windows command line because of the shebang support. Add **.py** and use `colctl.py` to support Windows file extension mappings. I am not so much into pip to give support here but afaik you should prefer `python -m pip` to use the correct version. Project files should all be in the cloned source. I am sure you will get into it quickly :-)

Interesting thing with the pump speed. Mine has never been so low with the X62. Sure you don't mean the fan? You talked about the fan with 60% earlier, where the minimum is 25%.

Carthago44, 2018-08-24:
Yes, I mean pump speed for sure, writing about fan was a mistake. I've now tried it several times and it works fine. Maybe it has to do with the kraken firmware - installing seems only to be possible via CAM.
Thank you for your tips so far; for the first I installed Python new with custom options in c:\programs and Path in system environment and I am reading a lot in the turtorial.

jneumaier, 2018-08-24:
Very nice. I read 60% would be a firmware limit. Seems to be 50% for me as well.

Carthago44, 2018-08-25:
My values reported by colctl actually:

**_>colctl -ps 50
>colctl.py -s
Device status:
fan_speed 514
liquid_temperature 34.4
pump_speed 1172
>colctl -ps 60
>colctl.py -s
Device status:
fan_speed 514
liquid_temperature 34.4
pump_speed 1759_**

jonasmalacofilho, 2018-08-25:
Can I ask what firmware version do you have on your Kraken?

Also, regarding setting speeds in general, can you check if you can reproduce #11 (wait 10+ seconds, `colctl -s` again)?

As for the minimum PWM duty, I had also noticed that the firmware would accept 50% (resulting in 1900-2000 rpm, normally).  I am not sure why CAM limits it to 60%, but there is the possibility that it is so for pump longevity: we only have the temperature of the liquid in a single spot, and at some low flow rates there could be significant differences.

jonasmalacofilho, 2018-08-26:
Thanks!

How fast does your pump spin at 100%?  And on load, how high does your liquid get?  Asking because ~1100 rpm does seem quite low.

But maybe it is just that the relationship between PWM duty and actual speed in rpm has also changed in recent firmware versions.

Carthago44, 2018-08-26:
To your question:
>colctl.py -ps 100
>colctl.py -s
Device status:
fan_speed 520
liquid_temperature 36.5
pump_speed 2826

Highest liquid temperature I recognized was about 45 degrees in days with a room temperature of nearly 30 degrees. CPU-temperature then was near 50.

Carthago44, 2018-08-26:
As an addition here the spec of x62 (in German only - sorry) where you can see the rpm range. So 1000 seems to be correct anyway.

Technische Daten:
Maße: Radiator: 315 x 143 x 30 mm (B x H x T)
Pumpe: 80 x 80 x 52,9 mm (B x H x T)
Gewicht: 1,29 kg
Materialien: Aluminium, Kupfer, Kunststoff, Gummi, Nylon
Pumpengeschwindigkeit: 1.000 - 2.800 U/Min +/- 300 U/Min
Spannung (Pumpe): 12 V
Anschluss (Pumpe): USB
LED-Beleuchtung (Pumpe): RGB
Schlauchlänge: 40 cm
Lüfter: 2x 140 mm (Aer P140)
Drehzahl: 500 - 1.800 U/Min +/- 300 U/Min)
Fördervolumen: 46,33 - 166,79 m³/h
Statischer Druck: 0,21- 2,71 mm H2O
Lautstärke: 21 - 38 dB(A)
Anschluss: 4-Pin-PWM
Kompatibilität: Intel-Sockel: 115x, 1366, 2011(-V3), 2066 | AMD-Sockel: AM2(+), AM3(+), AM4, TR4, FM1, FM2(+)
Herstellergarantie: 6 Jahre

jneumaier, 2018-08-26:
The initial issue should be fixed now. I will leave the issue open for a few days for additional questions or feedback.
We can discuss the fan and pump speed behaviour in #11.

jneumaier, 2018-08-26:
@Carthago44 I would really like my pump speed as low as yours though. I opened a new issue #20 for the 50% lower pump limit.

I also merged @jonasmalacofilho's contribution to read the firmware version with:
`colctl.py -s`
You have to update your source for this feature (`git pull`). Maybe you can verify the firmware version now.

jonasmalacofilho, 2018-08-27:
@Carthago44 I see different specs on NZXT's website, even in German.

Can you provide a link to those specs, or a picture in case it's something that came with your unit? 
Having this properly documented could prove very useful for the discussion in #20.

Carthago44, 2018-08-27:
@jonasmalacofilho : for example to find here: 
http://www.hardwareoverclock.com/NZXT-Kraken-X62-AIO-Wasserkuehlung.htm
I also got such specs in paper form with my x62, but I can't find it anymore…..

@jneumaier 
Thanks for new source - I will test it as soon as possible.
Trying to get familiar with python I try experiments like reading cpu temp and therefore would like to install pyspectator, but at the moment this ends with error messages:
>pip install pyspectator
Collecting pyspectator
  Using cached https://files.pythonhosted.org/packages/96/04/7ced429c14e26fe44d121ce4517b8894491f732b46e482d5067320b1e3b9/pyspectator-1.2.1.tar.gz
    Complete output from command python setup.py egg_info:
    error in pyspectator setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Invalid requirement, parse error at "'-r base.'"
  ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\Toni\AppData\Local\Temp\pip-install-elgo935w\pyspectator\

jneumaier, 2018-08-27:
@Carthago44 Seems to be a bug already fixed in the repository: https://github.com/it-geeks-club/pyspectator/issues/23

The issue has not been responded to, but the 1.2.1 release version includes the `-r base.txt` line:
https://github.com/it-geeks-club/pyspectator/blob/45160084b56067f2ce94d9d5c282fe11aa81e443/requirements/windows.txt
The current master version seems fixed:
https://github.com/it-geeks-club/pyspectator/blob/master/requirements/windows.txt

I guess you can install the pip package from source like you did for krakenx and test it. Maybe you can also comment their issue.

Carthago44, 2018-08-28:
@jneumaier Thank you very much for perfect support! 
Pyspectator is installed now (after required installation of visual studio 14).
Reading firmware version works now too.

closed 2018-09-05 by jneumaier