KsenijaS/krakenx #19: Allow additional maintainers / collaborators

jneumaier, 2018-08-26:
@KsenijaS: Would you please consider if you would accept additional maintainers / collaborators in this project?

On rule could be that anyone but you may not commit directly to this repository but must create a pull request reviewed by at least one of the other collaborators. It is not an important thing but I got the impression you currently cannot spare much time and I do not see much risk in trying it out. It would also be nice to allow additional PyPi pip package collaborators for package updates.

KsenijaS, 2018-08-26:
@jneumaier Thank you for taking such interest in this project, indeed I don't have enough time right now. Thank you also for helping others use it.
I updated pypi package and added you as a collaborator, I don't have your user name for pypi account (I suppose that is what I need to add you as maintainer)

jneumaier, 2018-08-26:
@KsenijaS Thank you very much for your openness and trust! I just registered on pypi.org as jneumaier.
Maybe you can update the github project description to "Python script to control NZXT cooler Kraken X52/X62/X72 in Linux and Windows" and add the tags "x72" and "windows". These updates I cannot do (and are you decisions anyway). This way the project would be found easier with the github search or other search engines :-)

closed 2018-09-05 by jneumaier