KsenijaS/krakenx #20: Lower pump speeds possible for X52/X62/X72?

jneumaier, 2018-08-26:
As discussed in #15 a lower pump speed may be possible.
We have three verified cases with 50% pump speed support for X62.
Can we at least verify this for one other device? Then we might change this limit.

jneumaier, 2018-08-26:
@Lyberta
Can you maybe try with X72 if you are interested?

For local installation of the patch execute:
```bash
git clone https://github.com/jneumaier/krakenx.git
cd krakenx/
git checkout enhancement-lower-pump-speeds
sudo python3 -m pip install -e .
```

To test execute:
```bash
colctl.py -ps 60
colctl.py -s
```
maybe wait some seconds (5 or so) and verify if you get a different/working result with
```bash
colctl.py -ps 50
colctl.py -s
```

To revert to pip package installtion execute (for Linux):
```bash
cd ..
rm -rf krakenx
sudo python3 -m pip uninstall krakenx
sudo python3 -m pip install krakenx
```

jonasmalacofilho, 2018-08-27:
I have been thinking about this for a while...

On one hand, the specs on NZXT's website state the 1,600~2,800 +/- 300 rpm range for the pumps in all 2nd gen. Kraken X coolers.  On the other, keeping pumps at 2100+ rpm might be a push for increased cooler longevity.

Maybe more importantly, unlocking 50% PWM could make [some pumps](https://github.com/KsenijaS/krakenx/issues/15#issuecomment-415955507) run way bellow the 1600 +/- 300 rpm.  Then again, the same user has reported a different version of the specification with a much lower minimum speed.

jonasmalacofilho, 2018-08-27:
By the way, I performed a few tests with my X62 unit.

At around 170 W of heat output I can only observe slight changes on CPU and liquid temperature (3°C and 1°C, respectively), between 50% PWM ~ 1950 rpm and 100% PWM ~ 2800 rpm.  Between 50% and 60%, differences are obviously smaller and only noticeable on the liquid temperature, even then still bellow 1°C.

The most perceptible impact is on pump noise, which is completely eliminated at 50% PWM, even when compared to virtually inaudible levels at 60% PWM ~ 2100 rpm (case open, ear close to the pump).  
Thus, on my particular setup it does seem that running the pump a bit slower would be beneficial.

What worries me is that there are several variables to account for, like peak heat output and variations on air temperature, plus the thermal characteristics of the interfaces between cooler and die.  Other users might experience more drastic changes, or might have less headroom to begin with.

---

Note: speeding up my pump causes it to emit a sound between clicking and gurgling; I am not sure what causes it, and I am not particularly concerned with it, but it could be something that also explains why I cannot see a lot of benefit from running it faster.

jneumaier, 2018-08-27:
I experience the same clicking and gurgling sometimes. Overall I am fine but would be nice to the pump even more silent.

I often read about those low pump speeds, some examples:
https://www.reddit.com/r/NZXT/comments/6lv0vm/pump_speed_on_kraken_x62/
https://support.camwebapp.com/forums/252256-cam-bugs/suggestions/33286240-kaken-x62-max-pump-rpm-depends-on-rgb-template
https://support.camwebapp.com/forums/252256-cam-bugs/suggestions/33581635-kraken-x62-firmware-update See comment below - "permanent boost to minimum pump speed"

Maybe they changed it for a reason. Best thing might be to create a development branch, then interested people can try it out from there for some time.

Lyberta, 2018-08-28:
X72 is working with 50 speed.

jneumaier, 2018-08-28:
@Lyberta Thank you!

Carthago44, 2018-08-28:
@Lyberta 
As long as usb is powered on the settings remain in kraken; after power down it starts with default values again.

jneumaier, 2018-09-05:
I added a [development](https://github.com/KsenijaS/krakenx/tree/development) branch with the 50% enhancement. Any feedback and additional tests are welcome. Maybe we can merge the behaviour when we feel more secure about the change.

jneumaier, 2018-09-17:
Merged current master into [development](https://github.com/KsenijaS/krakenx/tree/development) (includes temperature profiles).

Carthago44, 2018-09-18:
Hello Jens,

 

I was not at home for some weeks but I will continue testing soon.

 

Greetings,

Toni

jonasmalacofilho, 2018-09-26:
I initially had some concerns, but I agree that this is an improvement and that we shouldn't limit our users.

Maybe we could add some general information about the need to (properly) stress test the system, and not overload the cooler?

---

>The main issue is that while allowed by the firmware, such low speeds
>are not (currently) exposed in CAM.  This probably means that there are
>scenarios where correlation has been observed between pump speeds lower
>than 60% and increased failure rates.
>
>However, no public statement has been made, and there should also be
>plenty of scenarios where they are perfectly harmless, for the following
>reasons:
>
> - they are allowed by the firmware
> - the pump would still run well above its minimum specified speed (i.e.
>   1900+ rpm, when the specification lists 1600 ± 300 rpm)[1]
> - the performance differences between 1900 rpm (50%), 2100 rpm (60%) and
>   even 2800 rpm (100%) are very small, both on liquid and component
>   temperatures[2][3]
> - while DC controlled devices might stall at low voltages, these pumps
>   are controlled with (high-frequency) PWM, thus greatly reducing the
>   likelihood of this issue[4]
>
>For these reasons, my position is (finally) that we should educate the
>users, instead of limiting them.
>
>[1] https://web.archive.org/web/20180918051402/https://www.nzxt.com/products/kraken-x62
>[2] According to my own testing with Kraken X62 + i7 8700K + 170+ W
>[3] https://youtu.be/3B7ou8fOFfE?t=1m35s
>[4] https://youtu.be/Kc07JtDerUw

– From commit that enables this on the future driver ( jonasmalacofilho/liquidctl@651e5bbbf83436af8e868e59f4eaefc6d602f667)

jonasmalacofilho, 2019-03-13:
Closing as #39 has been merged.

commit 09c15b03 (2018-09-05): Fixed #20 support lower pump speeds (50%)
closed 2019-03-13 by jonasmalacofilho