KsenijaS/krakenx #21: Write only specified values for fan, pump and color schemes

jneumaier, 2018-08-27:
I would prefer a behaviour where only specified fan/pump profiles/values (profiles currently in development #11) are set if executing the colctl command. Currently the update method does it all with no conditions:

```python
self._send_color()
self._send_fan_speed()
self._send_pump_speed()
return self._receive_status()
```

I think we should change this and include checks for given command values. Addtionally we might also execute print_status by default instead of returning an ignored status value.

Feedback and pull requests welcome :-)

jneumaier, 2018-09-17:
Should be postponed until new liquidctl driver is included #11.

jonasmalacofilho, 2019-10-05:
Closing due to lack of activity.

closed 2019-10-05 by jonasmalacofilho