KsenijaS/krakenx #22: [feature request] option to disable logo on mode Spectrumwave

Lolzen, 2018-08-27:
An flag to either have a solid color, or none at all on the logo (NZXT text") while maintaining spectrumwave on the ring would be nice.

jneumaier, 2018-08-27:
Spectrum wave completely ignores all input colors in the protocol as far as I know and tested. See also [PROTOCOL.md](https://github.com/KsenijaS/krakenx/blob/master/PROTOCOL.md). Do you know if this is possible in CAM software?

Lolzen, 2018-08-27:
It is possiblle on CAM software, hence why i have issued a request for this.

jonasmalacofilho, 2018-08-27:
@jneumaier logo and ring can be set to different modes.

The third byte specifies which channel to target, sync both, logo only or ring only.

jneumaier, 2018-09-05:
I will try to fix this soon (hopefully next weekend). If someone starts working on this, please just leave a notice here.

jneumaier, 2018-09-05:
FYI see "only ring" flag and references in:
https://github.com/jonasmalacofilho/liquidctl/blob/f42213366a3bdca27e4ba66e8b5986f979f1eef3/liquidctl/driver/kraken_two.py#L51-L80

jonasmalacofilho, 2018-09-18:
We can defer this to #27, as it should come for free with the new driver.

In terms of UI, I think we can infer `logo` vs `ring` vs `sync` from the supplied options (`--text_color` vs `--color<n>`.  But this would depend on #21.

jonasmalacofilho, 2019-03-13:
Closing as #39 has been merged.

closed 2019-03-13 by jonasmalacofilho