KsenijaS/krakenx #25: Support for X42?

atyshka, 2018-09-14:
This repo doesn’t advertise support for the Kraken X42 but has anyone tried it? I’d like to have this control if possible but I don’t want to risk messing up my cooler

jonasmalacofilho, 2018-09-14:
The only way to know for sure is to test it (it is what we have done with our own coolers).  If you decide to give it a go, you might feel more comfortable to start with `colctl --status`, as it is a read-only operation.

Assuming that works, you can then try the settings that are expected to work with your cooler.  For example, avoid any color modes that you don't see in CAM.  And please report any features that either broken or missing.

One last thing you should be aware of is that, apart from the `--status`  and `--help` options, settings you leave unspecified will still be pushed to the device using hardcoded defaults.  From the top of my mind, those are solid red color, 30% fan speed and 60% pump speed.

atyshka, 2018-09-14:
Okay cool! I won’t be able to see what CAM supports as this is a Linux only computer, but I’ll be conservative. Also, just out of curiosity, does the cooler remember it’s settings or do I need to run this command after every boot?

jonasmalacofilho, 2018-09-14:
The cooler remembers the settings until it is completely cut from USB power (i.e. disconnected from the motherboard or the PSU switch is turned off).  So you might want to set your system to reconfigure itself at boot time.

jonasmalacofilho, 2018-09-14:
By the way, a sibling project indicates that it should indeed be compatible (see this [commit](https://github.com/brkalmar/leviathan/commit/0e22fb6ae783abc0746316b849c41ff5e178413f#diff-04c6e90faac2675aa89e2176d2eec7d8)).

jonasmalacofilho, 2018-09-14:
Actually, a [few commits were added](https://github.com/brkalmar/leviathan/issues/1) but, from a quick glance, they wouldn't really matter in the case of krakenx.

KsenijaS, 2018-09-14:
Hi atyshka,
It should remember the settings unless if it's in Solid/SolidAll mode, on my machine I have to rerun the command.

jonasmalacofilho, 2018-09-14:
@atyshka let us know how well it worked and if you encountered any issues.

jonasmalacofilho, 2018-10-15:
Closing due to inactivity.

I also think we can safely assume it works, based on `leviathan` and the absence of issue reports (there or here).

closed 2018-10-15 by jonasmalacofilho