KsenijaS/krakenx #26: Instructions for Windows users

jonasmalacofilho, 2018-09-15:
I think the Windows installation guide should also list a method that does not block CAM from accessing the device.

It would be nice if there was a user-friendly way of install the libusb 1.0 client libraries on Windows.  On liquidctl I had to settle for a manual guide to how they should be installed, but that is not optimal.

Anyone willing to write a script that does the necessary steps (download the appropriate release from libusb and copy the dlls to the correct places, taking into account the different ways Python might be installed)?

jneumaier, 2018-09-17:
I am a little confused about all these alternative libraries and version. For example is the libusb project from github much different from the libusb-win32 version? They also suggest Zadig for an easy installation process like we do here: https://github.com/libusb/libusb/wiki/Windows
Maybe it works only simultaneously with CAM as a filter driver installation instead of the replacement. Did not get this to work easily and I was happy to never see CAM again though :-)

jonasmalacofilho, 2018-09-17:
Libusb is just a cross-platform library to access USB devices.  On Windows, it is compatible with the generic HID device and some others; on Linux it always runs with its own driver, but there we also have detach/reattach APIs to return control after we're done with the device.  Oh, and libusb-win32 packages an older version of libusb (0.1).

The main issue I see with using the Zadig tool: it will only install the libraries in a user-friendly way **if** you force a driver change... but that also brings incompatibilities, most notably with CAM but also with other tools that might come up.

Ideally there would be an automated/easy way to install libsub _without touching the driver_.  Because, as far I can see, the generic HID driver supplied in Windows is the optimal one for us and our users: we don't use any features from the non-standard drivers.

P.S. liquidctl in particular is not affected with the choice of driver and libraries because it is compatible with most, if not all of what Zadig might install.

jneumaier, 2018-10-03:
Thanks for the info. I will have a look into this on the next weekend.

jonasmalacofilho, 2019-10-05:
Closing due to lack of activity.

commit e7f0370d (2020-07-31): Don't recommend the replacement of stock Windows drivers

This was never a good idea.  In fact, the recommendation was originally
introduced to, through a side-effect of how Zadig works, install libusb.

More importantly, liquidctl 1.4.0 will no longer support accessing HIDs
on Windows through non-HID drivers.

Remove the old recommendation, and link to instructions on how to
restore the stock Microsoft Generic HID driver.

Fixes: #26 ("Instructions for Windows users")
closed 2019-10-05 by jonasmalacofilho