KsenijaS/krakenx #27: Switch to the liquidctl driver

jonasmalacofilho, 2018-09-18:
We have decided in #11 to switch to the liquidctl driver.  I'm opening this issue so that we can track that effort.

Obviously we need to:

 - [ ] replace the current implementation and hook into the new driver

@KsenijaS has pointed out that we also need to:

 - [ ] update the README with the change and also new user-facing features
 - [ ] update the PROTOCOL documentation; @jneumaier has suggested to make it a summary of a larger document that will accompany the driver (or, until then, its code)

I also think we should do a couple of releases for PyPI:

 - [ ] a final `0.0.4` release with the current implementation
 - [ ] a `1.0.0` release to start a cycle with the new driver

jonasmalacofilho, 2018-09-18:
By the way, who of us will actually implement this? : )

I can do it of course, but maybe @jneumaier can take this on as well?

---

By the way, I will prepare a 1.1.0 release of liquidctl (with a few extra color modes and maybe some other small features), so that krakenx can just import it via setup.py/PyPI.

jonasmalacofilho, 2018-09-24:
I started working on the initial driver switch, see [jonasmalacofilho:switch-to-liquidctl-driver](https://github.com/jonasmalacofilho/krakenx/tree/switch-to-liquidctl-driver).

Note: you need a current 'master' version of liquidctl, while I finish preparations for the upcoming release.

jneumaier, 2018-09-24:
@jonasmalacofilho Sorry for the late reply. I hoped to find some time to help. To be honest it might have taken me quite some time to find this "easy solution" in the end. The code is really comprehensible now.

Integrates very well and works fine already. Only the logo/ring/sync separation needs some more work from what I tested. Maybe an easy solution would be to always set ring only and separately set text color if a value is given (except for MODE_SOLID_ALL). And "Write only specified values for fan, pump and color schemes #21" would be nice later on. Makes no sense that I always overwrite my profiles playing around with colors.

I would be fine to harmonize the mode names a little (Police, Spinner, Chaser) - no idea which names make more sense, guess the krakenx ones are made up? :-) Would make it easier to maintain the documentation (we could add aliases but mode name changes would be fine in my opinion).

jonasmalacofilho, 2018-09-26:
> Maybe an easy solution would be to always set ring only and separately set text color if a value is given.

That's a good idea.  But there are cases where syncing both channels might also be desired (e.g. breathing).

In the end, I think we really need a `--color_channel`/`-ch` option, and I just added that to the branch.

This should fix #22 and allow us to enable all other remaining color modes.  But, at the same time, we are already somewhat moving towards the adoption of #21: two separate calls would be necessary in those cases, in this changes how the UI is used.  But I'm fine with it... I agree with you that having it

> overwrite my profiles playing around with colors"

is not optimal.  Also, I change fan and pump speeds often; I don't want to think about the RGB all the time.

jneumaier, 2018-10-03:
I moved protocol.md and added a note about the driver in there. For now I would be fine with this. You may add further adjustments to the documentation or just mark the task as done :-)

jonasmalacofilho, 2018-10-03:
@jneumaier thanks!

jonasmalacofilho, 2019-01-13:
Rebased and updated [jonasmalacofilho/switch-to-liquidctl-driver](https://github.com/jonasmalacofilho/krakenx/tree/switch-to-liquidctl-driver) ([diff](https://github.com/KsenijaS/krakenx/compare/master...jonasmalacofilho:switch-to-liquidctl-driver?expand=1)):

 - require liquidctl v1.1.0
    + closes #29
    + incorporates #20
 - add all missing base lighting modes
    + includes: Off, CustomWave, Alternating, MovingAlternating, CustomBreathing, Loading
    + missing: backward variants; should I just follow liquidctl (e.g. BackwardsMovingAlternating)?
    + missing: different Marquee lengths (3 < length <= 6)
 - handle channels with `--color_channel`/`-ch`
    + closes #22

@jneumaier can you take a look on the naming and missing documentation?

KsenijaS, 2019-01-13:
Hi Jonas,

Thank you for making those changes! They will make the code much more readable. I have a few comments though:

1. I wouldn't replace MODE_SOLID with COLOR_MODES[1]; line 44 is not readable at all. It's not obvious what COLOR_MODES[1] is, nor is it obvious in COLOR_MODES's definition that the order of modes would change what the code does. Maybe consider defining a MODE_SOLID mode outside COLOR_MODES, inserting it to COLOR_MODES at the right index, and referring to it directly?

2. Using the string "super-" to denote that text color can be changed on a mode is error prone and unreadable. First, it's not obvious in a mode's definition that prefixing its name with "super-" does anything functional. Second, it's not obvious what "super-" means. I think it'd be better if this were a third field in the mode namedtuple, and if it were called something more declarative (e.g. custom_text_color or something)

I wanna hear what you think.

jonasmalacofilho, 2019-01-14:
Thanks, you were right, those were ugly.

1. Did basically that.

2. While `super` does mean _something_ in the driver – a mode where LEDs are referenced individually – the code was indeed unnecessarily convoluted.  Fixed that by adding a new `uses_text_color` field to the `Mode` tuple.

By the way, while strange, most modes don't actually do anything with the supplied text color.  IIRC _one_ other (non "super") mode used it, but I decided to hid that in liquidctl/the driver and keep things simple.  CAM doesn't expose that either, and selecting the appropriate channel will always give better results.

jneumaier, 2019-01-26:
Thanks a lot @jonasmalacofilho 

How should we proceed here?

- [ ] @jneumaier: Test switch-to-liquidctl-driver against current README.md documentation and create a pull request against jonasmalacofilho/switch-to-liquidctl-driver for changes
- [ ] @jneumaier: Add some Windows documentation #26 including suggestions for autostart on login
- [ ] Merge into master
- [ ] Optional: (I can do this) Add pull request for #21 
- [ ] Release 1.0.0

I have a lot of time at the first weekend in February and will finally help a little here. Sorry for the delay.

Feel free to edit this post.

PR #39 (2019-03-13): Switch underlying implementation to a liquidctl driver
In [#11](https://github.com/KsenijaS/krakenx/issues/11#issuecomment-421105103), and later in #27, we have been discussing and working on reducing the duplicated efforts by concentrating the low level protocol stuff in liquidctl.

These changes modify the `KrakenX52` device class to use `liquidctl.driver.KrakenTwoDriver` under the hood.  Argument parsing and validation is kept, as krakenx takes a different and more explicit approach to this.

I'm opening the pull request to get things moving again.  It also makes these changes easier to fin [... cut]
closed 2019-03-13 by jneumaier