KsenijaS/krakenx #28: Steps to AutoRun Script Upon User Login

kkobashi, 2018-09-28:
Here are instructions on how to auto run this script to make sure the fans are running. The script gets executed after the user logs in. This is very important if you do intensive computing running the CPU hot. 
Unlike Windows where the AIO automatically kicks in via Cam, this won't happen under Linux.

Below was executed on Ubuntu 18.0.4.

### Get Python and PIP

sudo apt install python3
sudo apt-get install python3-pip
mkdir /home/username/Apps
cd /home/username/Apps
mkdir krakenx

### Download krakenx off GitHub

Copy zip to /home/username/Apps/krakenx
unzip krakenx-master.zip
cd krakenx-master
python3 -m pip install krakenx (don't sudo or file error with owner cache permission appears)

### Create a startup.sh

cd /home/username/Apps/krakenx/krakenx-master/bin
sudo nano startup.sh

```
# Must change dir into location of Python script colctl
cd /home/username/Apps/krakenx/krakenx-master/bin

#Must sudo or the usb driver will have access permission problems
sudo python3 colctl --mode SpectrumWave -fan_speed "(20,25),(30,60),(40,80),(50,100)"
```

### Set Permissions and Ownership of startup script

sudo chown username:username 755 startup.sh
sudo chmod 755 startup.sh

### Run script upon user logging in

cd ~
sudo nano .profile

`/home/username/Apps/krakenx/krakenx-master/bin/startup.sh`

Note that you can have separate profiles and separate colctl settings per user.

Hope this helps someone.

jonasmalacofilho, 2018-09-28:
Thanks!

This should be included in the documentation.

---

@jneumaier where do we want to put the growing documentation: `README` only, `/docs` or Wiki?

jneumaier, 2018-09-29:
@kkobashi Thank you!

Good idea, I guess everyone needs this somehow. I might add a Windows version using the task scheduler.

@jonasmalacofilho I would prefer to split the extended documentation in some extra place and link it from `README.` Maybe a wiki would be a clear favorite for a bigger number of documents. In our case I would prefer a `/docs` directory but I am fine with both.

kkobashi, 2018-10-02:
I've updated this to use a system daemon instead. There's too much to consider regarding interactive, non-interactive, non-login, login cases and the order the scripts are executed. Cron upon reboot, even every 1 minute didn't trigger colctl. I think it may have to do with udev rules, USB, and permissions.

jneumaier, 2018-10-03:
@kkobashi I integrated you documentation. Please feel free to reopen this issue or create a pull request any time.

kkobashi, 2018-10-09:
@jneumaier When you copy and pasted, the formatting got messed up. I tried fixing but I'm prevented from doing so. Please fix so its readable in the same format as I have here. In particular, the commands are all run together on one line. They need CR/LF probably as they were probably stripped by your Windows editor.

jneumaier, 2018-10-09:
@kkobashi Git manages line endings even if Windows users mess them up :stuck_out_tongue: They were all fine, but I thought it was valid markdown syntax already which handles breaks automaticallly. Sorry, should have checked this more thorough.

jonasmalacofilho had some suggestions for improvements he left in commit comments, see https://github.com/KsenijaS/krakenx/commit/6bf4f925a187e4b51df29b81cd5da621a37d9b29
You can always create a pull request from your own fork:
https://help.github.com/articles/creating-a-pull-request-from-a-fork/

jneumaier, 2018-10-09:
We should reference the default installation procedure in [doc/linux-autorun.md](https://github.com/KsenijaS/krakenx/blob/master/doc/linux-autorun.md). Otherwise this will be hard to maintain.

Can we not just install as described in README.md and do
`ExecStart=colctl`
Don't we have a PATH resolval here? What is the reason for this custom installation procedure?

jonasmalacofilho, 2018-10-09:
There are a few quirks – e.g. systemd units don't see the same environment (variables) as you do – but it can usually be managed without installing things twice or requiring them to be in `/etc`.

Though I haven't tested this with krakenx,[1] and @kkobashi also mentions a possible (and likely) dependency issue with udev.

[1] I'm traveling for work, I'm not sure I'll be able to test this before I get back.

closed 2018-12-05 by jonasmalacofilho

jonasmalacofilho, 2018-12-05:
Closing, as there is nothing to be done _directly_.

I've been testing a [simpler](https://github.com/jonasmalacofilho/dotfiles/tree/master/krakenx) setup.  Eventually, I'll _propose_ a complete rewrite of the instructions based on this simpler approach, as the custom installation (and inside `/etc`, of all places) really does not sit well with me.

Note: systemd 239.3030-1 introduced [some](https://github.com/jonasmalacofilho/dotfiles/commit/e721bfeb36590f2b11bc2bea7d537e607e2c634e#diff-e97b0c0348098daf807e74b3ad362bb8) issue related to the initialization of the device (not dbus).  I already researched how to solve that [properly](https://lists.fedoraproject.org/pipermail/devel/2012-January/160917.html), but haven't yet had the time to test it.

commit 6bf4f925 (2018-10-03): Resolve #28 add linux autorun how-to
Split documentation to separate folder suggested and update protocol.md to reflect #27
commit a0600b36 (2018-10-03): Resolve #28 change personalized comments
commit e8a157d6 (2018-10-09): Resolve #28 fix markdown syntax