KsenijaS/krakenx #29: "Frozen" values when reading status after changing FS and/or PS

Carthago44, 2018-10-01:
The values of PS and FS shown by -s are only correct up to the first change via -ps xx and/or -fs xx.
Afterwards reading status gives me the correct value of the liquid temperature, but FS and PS always show the same. Physically the fan (and the pump too I hope - cannot hear it, because changing only ps also alters fs(?)) reacts correct.

jonasmalacofilho, 2018-10-01:
Can you confirm which version of krakenx you are on?

Also, what the firmware version (`colctl -s` should show that) do you have
again?

Carthago44, 2018-10-01:
My firmware is 2.1.8. This was the last version which I could install with CAM before changing the driver for krakenx.

Yes, I mean -ps <value1> -fs <constant> and -ps <value2> -fs <constant> result in different fan speeds.

jonasmalacofilho, 2018-10-01:
> Yes, I mean -ps -fs and -ps -fs result in different fan speeds.

What krakenx version do you have?

If you are not already on the latest v0.0.3, can you that or what's currently on `master`?

Carthago44, 2018-10-01:
Seems I have some problems writing text here …
I tried to quote your sentence, but didn't work - sorry!

Carthago44, 2018-10-01:
I have an X62 and I use the development branch 0.0.3

jonasmalacofilho, 2018-10-01:
> I tried to quote your sentence, but didn't work - sorry!

No worries!

>I use the development branch 0.0.3

Thanks!

---

Well, this is a strange issue.

Can you run the same test on a [version with some debug messages](https://github.com/jonasmalacofilho/krakenx/tree/debug-29) and send us the output?

Carthago44, 2018-10-01:
Ok, done, lots of text now, hope it helps you:

Device status:
received 04 21 03 02 07 04 a5 00 00 00 40 02 00 01 08 1e 00
fan_speed 519
firmware_version 2.1.8
liquid_temperature 33.3
pump_speed 1189

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 60 -ps 70
sending 02 4c 00 00 00 00 ff 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 (+33 padding bytes)
sending 02 4d 80 14 3c (+60 padding bytes)
sending 02 4d 81 16 3c (+60 padding bytes)
sending 02 4d 82 18 3c (+60 padding bytes)
sending 02 4d 83 1a 3c (+60 padding bytes)
sending 02 4d 84 1c 3c (+60 padding bytes)
sending 02 4d 85 1e 3c (+60 padding bytes)
sending 02 4d 86 20 3c (+60 padding bytes)
sending 02 4d 87 22 3c (+60 padding bytes)
sending 02 4d 88 24 3c (+60 padding bytes)
sending 02 4d 89 26 3c (+60 padding bytes)
sending 02 4d 8a 28 3c (+60 padding bytes)
sending 02 4d 8b 2a 3c (+60 padding bytes)
sending 02 4d 8c 2c 3c (+60 padding bytes)
sending 02 4d 8d 2e 3c (+60 padding bytes)
sending 02 4d 8e 30 3c (+60 padding bytes)
sending 02 4d 8f 32 3c (+60 padding bytes)
sending 02 4d 90 34 3c (+60 padding bytes)
sending 02 4d 91 36 3c (+60 padding bytes)
sending 02 4d 92 38 3c (+60 padding bytes)
sending 02 4d 93 3a 3c (+60 padding bytes)
sending 02 4d 94 3c 64 (+60 padding bytes)
sending 02 4d c0 14 46 (+60 padding bytes)
sending 02 4d c1 16 46 (+60 padding bytes)
sending 02 4d c2 18 46 (+60 padding bytes)
sending 02 4d c3 1a 46 (+60 padding bytes)
sending 02 4d c4 1c 46 (+60 padding bytes)
sending 02 4d c5 1e 46 (+60 padding bytes)
sending 02 4d c6 20 46 (+60 padding bytes)
sending 02 4d c7 22 46 (+60 padding bytes)
sending 02 4d c8 24 46 (+60 padding bytes)
sending 02 4d c9 26 46 (+60 padding bytes)
sending 02 4d ca 28 46 (+60 padding bytes)
sending 02 4d cb 2a 46 (+60 padding bytes)
sending 02 4d cc 2c 46 (+60 padding bytes)
sending 02 4d cd 2e 46 (+60 padding bytes)
sending 02 4d ce 30 46 (+60 padding bytes)
sending 02 4d cf 32 46 (+60 padding bytes)
sending 02 4d d0 34 46 (+60 padding bytes)
sending 02 4d d1 36 46 (+60 padding bytes)
sending 02 4d d2 38 46 (+60 padding bytes)
sending 02 4d d3 3a 46 (+60 padding bytes)
sending 02 4d d4 3c 64 (+60 padding bytes)
received 04 21 01 02 07 04 a5 00 00 00 40 02 00 01 08 1e 00

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
Device status:
received 04 21 01 02 07 04 a5 00 00 00 40 02 00 01 08 1e 00
fan_speed 519
firmware_version 2.1.8
liquid_temperature 33.1
pump_speed 1189

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 80 -ps 80
sending 02 4c 00 00 00 00 ff 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 ff 00 00 (+33 padding bytes)
sending 02 4d 80 14 50 (+60 padding bytes)
sending 02 4d 81 16 50 (+60 padding bytes)
sending 02 4d 82 18 50 (+60 padding bytes)
sending 02 4d 83 1a 50 (+60 padding bytes)
sending 02 4d 84 1c 50 (+60 padding bytes)
[... 77 lines cut ...]
sending 02 4d c7 22 32 (+60 padding bytes)
sending 02 4d c8 24 32 (+60 padding bytes)
sending 02 4d c9 26 32 (+60 padding bytes)
sending 02 4d ca 28 32 (+60 padding bytes)
sending 02 4d cb 2a 32 (+60 padding bytes)
sending 02 4d cc 2c 32 (+60 padding bytes)
sending 02 4d cd 2e 32 (+60 padding bytes)
sending 02 4d ce 30 32 (+60 padding bytes)
sending 02 4d cf 32 32 (+60 padding bytes)
sending 02 4d d0 34 32 (+60 padding bytes)
sending 02 4d d1 36 32 (+60 padding bytes)
sending 02 4d d2 38 32 (+60 padding bytes)
sending 02 4d d3 3a 32 (+60 padding bytes)
sending 02 4d d4 3c 64 (+60 padding bytes)
received 04 20 06 02 07 04 a5 00 00 00 40 02 00 01 08 1e 00

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
Device status:
received 04 20 06 02 07 04 a5 00 00 00 40 02 00 01 08 1e 00
fan_speed 519
firmware_version 2.1.8
liquid_temperature 32.6
pump_speed 1189

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>

jonasmalacofilho, 2018-10-01:
Do you remember the last krakenx version where this used to work?

Have you tried turning off the PSU long enough for the device to reset to its defaults?

---

Ideas I'm considering:

 - fan and/or pump profiles not supported on 2.1.8?
 - fan and/or pump profiles supported, but have limitations on 2.1.8?
 - another krakenx implementation issue?
 - driver issue?
 - controller has crashed?
 - unit issue?

Carthago44, 2018-10-01:
Sorry, no idea to a working version number.
I bought my kraken in July, started with CAM and the well-known problems and dicovered krakenx at the beginning of August. Python and GitHub was quite new for me...…

Turning power off over night should be long enough for kraken to forget everything I hope.

jonasmalacofilho, 2018-10-01:
> Sorry, no idea to a working version number.

But it used to work, right?  When we were discussing lower pump speeds in #15?

> Turning power off over night should be long enough for kraken to forget everything I hope.

More than enough!  In just a couple minutes all capacitors should have discharged fully and the cooler will reset to its default settings (including loading/white mode/color).

Carthago44, 2018-10-01:
Yes, it used to work. Otherwise I wouldn't have been able to report my values.

jonasmalacofilho, 2018-10-01:
It's probably an issue with the fan/pump profiles; more specifically, with using 2 °C increments instead of the default (and not that useful) 5 °C steps.  If I'm this will be easy to avoid (though it won't perform as well as it could)... I'll implement something later today or tomorrow.

I would also like to see if there's a way to select the appropriate behavior without waiting for a read to get the firmware version.  If you don't mind, I'll include some other debug information in the next test to figure this out.

Carthago44, 2018-10-01:
Many thanks for your commitment here!
Of course I will try to help here with testing and so on. Unfortunately python is still very "new" for me and I try to learn, but time is rare......

Carthago44, 2018-10-02:
Another "nice" effect:
sending -ps xx and/or -fs xx results in an error message from my BIOS when I perform a warm start - tells me "CPU Fan Error", what means, that there is no more PWM signal from x62. After power off and on the signal is back and everything seems to work quite normal.
I didn't notice this behaviour up to now, because normally I switch power off after shut down.

jonasmalacofilho, 2018-10-02:
Wow!

But when you say everything is normal now, do you mean that even colctr/krakenx is now functioning too (even multiple runs)?

Carthago44, 2018-10-02:
No, this behavior seems to be reproduceable. 
As long as I use only -s to read the actual values everything works correct after new start without power off.  Using -ps xx -or fs xx brings back the problem - PWM off!

jonasmalacofilho, 2018-10-02:
Can you test the latest version on that same branch ([jonasmalacofilho/debug-29](https://github.com/jonasmalacofilho/krakenx/tree/debug-29))?

I would suggest starting from a (mechanical/complete) power off and trying out a few things.  Let me know what speeds are reported between each test and, also, if what you can hear from the fans and pump matches what's being reported.

```
colctr -s
colctr -fs 25 -ps 50
colctr -fs 100 -ps 50
colctr -fs 25 -ps 100
colctr -fs 50 -ps 100
```

If this doesn't work, the next logical step will be to assume that profiles are not supported at all on that firmware version.

Also, are setting any lighting modes?

---

Changes in the version you will run:

 - switch both profiles to use 5°C steps
 - trace more information to identify the device version, other than the firmware
 - disable read/write traces

Carthago44, 2018-10-02:
Ok, here the results:

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"
Device status:
fan_speed 518
firmware_version 2.1.8
liquid_temperature 32.7
pump_speed 1189

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 25 -ps 50
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"

---no changes of fan or pump to hear

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"
Device status:
fan_speed 533
firmware_version 2.1.8
liquid_temperature 32.7
pump_speed 1240

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 100 -ps 50
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"

---full speed fan sound

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"
Device status:
fan_speed 533
firmware_version 2.1.8
liquid_temperature 32.7
pump_speed 1240

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 25 -ps 100
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"

---- fan noise reduces to start values, can't hear any difference from pump

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"
Device status:
fan_speed 533
firmware_version 2.1.8
liquid_temperature 32.5
pump_speed 1240

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -fs 50 -ps 100
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"

--- fan noise increses a little bit, nothing to hear from pump

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>colctl -s
debug: bcdDevice=0x200 manufacturer="NZXT.-Inc." product="NZXT USB Device"
Device status:
fan_speed 533
firmware_version 2.1.8
liquid_temperature 32.1
pump_speed 1240

C:\Users\Toni\Documents\GitHub\krakenx-debug-29>

No, I don't use the light modes. My case is completely closed.

Carthago44, 2018-10-02:
Addition:

PWM of x62 is off again after this test!

jonasmalacofilho, 2018-10-02:
Thank you again!

I guess no profiles are supported at all[1] in 2.1.8.  A conditional check is needed for this.

Just so I don't have to deal with this twice, I've just pushed a temporary (non-conditional, just for you/2.1.8) quick fix to the branch you are testing...  I'll work on a more thorough solution in #27.

_Note: if you give it a profile, it will simply pick the first speed it sees._

As usual, let me know if this solves the problem or if you observe any other issues.

---

On a different front, it you want to update the firmware to use the profiles, you would only need to [uninstall the Zadig driver](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it).

To be honest, Zadig isn't strictly necessary as long as you install libusb by some other means.  We only recommend it for it's simplicity of installation (see #26).  By reverting to the Microsoft HID driver you could not only update the firmware but also see it in HWInfo64 (beta).  If you decide to go this route:

> libusb v1.0.21 is recommended; later versions can crash when trying to release the device.

> A simple way of installing it is to download the appropriate package from [libusb/releases](https://github.com/libusb/libusb/releases) and extract the `.dll` and `.lib` files that match you runtime (e.g. MS64) to your python installation directory (e.g. `%homepath%\Anaconda3\`).

---

[1] Actually, we're not completely crashing the controller as at least the fan speed keeps responding to our commands.  This makes me think that we could be able to get the profiles to work, at least in some cases or for only the fan channel.  However, this would require a lot of testing and, as a result/preferably, direct access to a cooler with this firmware version.

jonasmalacofilho, 2018-10-02:
Note: forgot to commit & push.  Just did that now.

Carthago44, 2018-10-02:
Thank you very much for the moment; I will continue testing tomorrow (11.45 pm here).

Carthago44, 2018-10-03:
Libusb is installed now - both colctl (your last debug version for me) and CAM work fine, but CAM software does not offer me a newer firmware version than 2.1.8.

jonasmalacofilho, 2018-10-03:
Ok, thanks for the update.  And it's too bad you can't update the firmware.

I'll let you know when a more permanent fix is merged.

PR #39 (2019-03-13): Switch underlying implementation to a liquidctl driver
In [#11](https://github.com/KsenijaS/krakenx/issues/11#issuecomment-421105103), and later in #27, we have been discussing and working on reducing the duplicated efforts by concentrating the low level protocol stuff in liquidctl.

These changes modify the `KrakenX52` device class to use `liquidctl.driver.KrakenTwoDriver` under the hood.  Argument parsing and validation is kept, as krakenx takes a different and more explicit approach to this.

I'm opening the pull request to get things moving again.  It also makes these changes easier to fin [... cut]
closed 2019-03-13 by jneumaier