KsenijaS/krakenx #30: Udev rule to remove the need for sudo on Linux

leinardi, 2018-10-02:
I just want to suggest the use of a `udev` rule to avoid the need of having to use `sudo` to execute `colctl`.

The steps necessary to achieve this are:

1. Creating the file `/lib/udev/rules.d/60-krakenx.rules` containing the rule `SUBSYSTEM=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"`

leinardi, 2018-10-02:
An example on how to do it automatically can be found here: https://github.com/flozz/rivalcfg/blob/master/setup.py

jonasmalacofilho, 2019-10-05:
Closing due to lack of activity.

And, IMO, quietly installing such udev rules in setup.py is bad: the system administrator should explicitly choose to allow access to unprivileged users.

closed 2019-10-05 by jonasmalacofilho