KsenijaS/krakenx #31: PEP8, PyLint and Mypy

leinardi, 2018-10-02:
I was checking the code and I noticed several violations of the PEP8, PyLint and Mypy rules. Would you be interested in a pull request that address this issues? If you are let me know and I will be happy to make one.

KsenijaS, 2018-10-02:
Hello Roberto,
Thank you for taking an interest in this project.
I think it would be great if you could make pull request.

Thanks

leinardi, 2018-10-02:
Sure, I will try to fix all the issues tomorrow.

What is the minimum python version you want to support?

KsenijaS, 2018-10-02:
I think minimum supported python versions should be 3.6 and 2.7

jneumaier, 2018-10-03:
@leinardi Please consider #27 and the major rewrite the project is currently implementing. I think you should not put to much effort in the driver itself because it will be switched soon and please also consider the current implementation from @jonasmalacofilho at [jonasmalacofilho:switch-to-liquidctl-driver](https://github.com/jonasmalacofilho/krakenx/tree/switch-to-liquidctl-driver). Maybe you can check with @jonasmalacofilho to avoid a lot of merge conflicts or do the pull request against his branch. Thanks for the initiative!

leinardi, 2018-10-03:
Yeah, better to wait than, otherwise there will be many conflicts.

jneumaier, 2019-01-26:
@leinardi 
You might already work on this and create a pull request against [jonasmalacofilho/switch-to-liquidctl-driver](https://github.com/jonasmalacofilho/krakenx/tree/switch-to-liquidctl-driver). We might release 1.0.0 milestone from this pull request but we can also finish this task afterwards.

jonasmalacofilho, 2019-10-05:
Closing due to lack of activity.

closed 2019-10-05 by jonasmalacofilho