KsenijaS/krakenx #32: usb.core.USBError: [Errno 110] Operation timed out

leinardi, 2018-10-02:
If multiple executions of `colctl` are done in a short amount of time this error is shown:

```
$ colctl --status
Device status:
Traceback (most recent call last):
  File "/usr/local/bin/colctl", line 121, in <module>
    main(parser)
  File "/usr/local/bin/colctl", line 115, in main
    cooler.print_status()
  File "/usr/local/lib/python2.7/dist-packages/krakenx/color_change.py", line 145, in print_status
    for k,v in sorted(self._receive_status().items()):
  File "/usr/local/lib/python2.7/dist-packages/krakenx/color_change.py", line 149, in _receive_status
    raw_status = self.dev.read(0x81, 64)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 851, in intr_read
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
```
Most probably the USB devices is not reset before

jneumaier, 2018-10-03:
Please also consider Windows support. It is broken now. I will fix this tommorow.

leinardi, 2018-10-03:
Sorry about Windows, I have not thought it could break it.