KsenijaS/krakenx #36: How to set control modes

varomix, 2018-12-04:
Hi
I think my fans are running a bit to hi, I can hear the the fans are a bit loud, maybe is in a Performance mode.
Is there a way to change Control Modes as described in the Website
[image]

I would love to have it in SIlent mode when I'm just doing basic stuff and Performance only when I need to.

thank you

jonasmalacofilho, 2018-12-04:
What `fan_speed` and `pump_speed` options have you used?

varomix, 2018-12-04:
think I'm at defaults cause I turned off everything and disconnected the cooler usbs, as someone said I could reset it.

varomix, 2018-12-05:
mmmh? I really don't want to get something wrong and kill my cooler
I guess I'll have to install windows and see if that has more control and try to figure it out on linux

thanks

jonasmalacofilho, 2018-12-05:
You're welcome.

I'll close this issue, but feel free to ask more questions.

closed 2018-12-05 by jonasmalacofilho