KsenijaS/krakenx #43: Not able to change text color in marquee modes

KhanSW, 2020-02-22:
I've been attempting to get a marquee animation going with matching text color (0,128,255) but no matter how I enter it I always get the color I want on the marquee but not the text (keeps defaulting to a pinkish / purple color).

The commands I have tried are:

colctl --mode marquee -c 0,128,255 -c0 0,128,255 -as 1
or
colctl --mode marquee --text_color 0,128,255 --color1 0,128,255 -as 1

When I try and add a color count to the argument it bypasses the color selection completely and gives me a red marquee with a teal text (no matter what rgb color code I enter).

Am I missing something?

jneumaier, 2020-02-22:
REAMDE.md says "Text color is always black" and it behaves the same for me.

You might want to try liquidctl:
https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md

It is a successor of this project and used as a driver for krakenx as well. If it is supported, it should work there. I should put a link to liquidctl on the main page :-)

KhanSW, 2020-02-22:
Thank you, that fixed it for me :)

closed 2020-02-22 by KhanSW